ArduPilot/MissionPlanner #3061: Drone disarms immediately after arm

mosheFuks, 2023-02-23:
Issue details
When the Arm/Disarm button is priced, the drone is armed, but after two seconds, it disarms. I'm using Ardupilot with AirSim. Any idea what can be happening?

Version
Mission Planner V4.4.0dev(https://github.com/ArduPilot/ardupilot/commit/b0f0733ec342657d3b9d85d991f8ca3dfb228c2a)

Platform
[ ] All
[ ] AntennaTracker
[ x ] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

Airframe type
Quad

Hardware type
SITL

mosheFuks, 2023-03-02:
I noticed that the Unhealthy GPS Signal message is displayed, could it be related to the mentioned error?

robertlong13, 2023-03-02:
This is a user support question, and should be posted on https://discuss.ardupilot.org/

When you do so, please provide a log.