ArduPilot/MissionPlanner #3077: Beta  will not update on PC from 1.3.79.1

Hwurzburg, 2023-03-12:
C:\Program Files (x86)\Mission Planner\Interfaces.dll.new Failed.
System.IO.IOException: Cannot create a file when that file already exists.

   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.File.InternalMove(String sourceFileName, String destFileName, Boolean checkHost)
   at Updater.Program.UpdateFiles(String directory) in C:\Users\mich1\Desktop\CubePilot\MissionPlanner\Updater\Program.cs:line 142
Move: C:\Program Files (x86)\Mission Planner\Interfaces.dll.new TO C:\Program Files (x86)\Mission Planner\Interfaces.dllC:\Program Files (x86)\Mission Planner\Interfaces.dll.new Failed.
System.IO.IOException: Cannot create a file when that file already exists.

   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.File.InternalMove(String sourceFileName, String destFileName, Boolean checkHost)
   at Updater.Program.UpdateFiles(String directory) in C:\Users\mich1\Desktop\CubePilot\MissionPlanner\Updater\Program.cs:line 142
Move: C:\Program Files (x86)\Mission Planner\Interfaces.dll.new TO C:\Program Files (x86)\Mission Planner\Interfaces.dllC:\Program Files (x86)\Mission Planner\Interfaces.dll.new Failed.
System.IO.IOException: Cannot create a file when that file already exists.

   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.File.InternalMove(String sourceFileName, String destFileName, Boolean checkHost)
   at Updater.Program.UpdateFiles(String directory) in C:\Users\mich1\Desktop\CubePilot\MissionPlanner\Updater\Program.cs:line 142

4O6FPV, 2023-03-12:
same problem even after a uninstall and reinstall.

robertlong13, 2023-03-13:
I'm also having trouble, but my error messages are different.

```
System.IO.IOException: The process cannot access the file because it is being used by another process.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.File.InternalMove(String sourceFileName, String destFileName, Boolean checkHost)
   at Updater.Program.UpdateFiles(String directory) in C:\Users\mich1\Desktop\CubePilot\MissionPlanner\Updater\Program.cs:line 142
```

I'm noticing that Planner isn't quitting when it's supposed to. The console prints
```
INFO MissionPlanner.Utilities.Update - Quitting existing process
INFO MissionPlanner.Controls.ProgressReporterDialogue - DoWork Done
```

If I manually close Planner at that point, I can get the updater to finish for me. Not sure why this is happening all of a sudden.