ArduPilot/MissionPlanner #3083: Data: Set EKF Origin Here scales altitude incorrectly

rmackay9, 2023-03-17:
MP's Data screen's right-mouse-button menu, "Set Home Here" uses the incorrect scale for the altitude.  It seems to be 100x too small.

The issue can be reproduced by running ArduPilot in SITL and doing the following:

- Start MP's simulator
- On the Data wscreen's Quick view ensure "alt", "altasl" and "terr_alt" are displayed
- On the Config, Full Parameter Tree screen, set
  - GPS_TYPE = 0
  - EK3_SRC1_POSXY = 0 (was 3)
  - EK3_SRC1_VELXY = 0 (was 3)
  - EK3_SRC1_POSZ = 0 (was 3)
- Restart MP and the simulator (you may check the "Skip Download" to save some time)
- Right-mouse-button-click on the map and select "Set Home Here" >> "Set EKF Origin HERE".  Note the "altasl" (e.g. alt above sea level) is about 100x lower than the terrain altitude shown on the Plan screen's SRTM field for the same location''
[image]
- Right-mouse-button-click on the map within a few meters of the vehicle and select "Set Home Here" >> "Set Home Here" and reply "Yes" to "This will reset the onboard home position" message.  Note how the "alt" (e.g. alt above home) is a large negative number.
[image]

Sadly this contributed to my copter with a ModalAI camera attached to destroy itself against the ceiling.

meee1, 2023-03-17:
please confirm your home location on the flight planner screen, including alt, and then start sitl again

casrya, 2023-03-29:
I took a look at the source code for both Mission planner and ArduPilot:

The mavlink message expects mm above sea level (see mavlink_set_gps_global_origin.h)
	int32_t altitude; /*< [mm] Altitude (MSL). Positive for up.*/

Ardupilot divides by 10 for cm in GCS_MAVLINK::handle_set_gps_global_origin()

Mission Planner sends meters above sea level from srtm.getAltitude(...) in setEKFHomeHereToolStripMenuItem_Click() so that is the problem. Interestingly DO_SET_HOME takes m so the confusion is understandable.

BTW @rmackay9, what is the use case for setting the EKF home manually? I have never used it but since it caused a vehicle crash I figured I should take a look at it!

rmackay9, 2023-05-16:
@casrya,

Sorry, I missed your message but thanks for providing advice on the fix.

The purpose of setting the EKF origin is for [non-GPS navigation](https://ardupilot.org/copter/docs/common-non-gps-navigation-landing-page.html).  So for example the ModalAI VOXL cameras or even simpler optical flow sensors.  If there's no GPS then AP doesn't know where the vehicle is (in lat, lon, alt terms) and this means it can't accept position targets in lat,lon,alt terms (e.g. waypoints).  If we added support in AP for specifying waypoints relative to the EKF origin then we could potentially get away without asking the user to specify where the vehicle is on Earth.

closed 2023-05-16 by rmackay9