ArduPilot/MissionPlanner #3092: UBUNTU: 22.04 does not show QUICK.

muramura, 2023-04-15:
#### Issue details
UBUNTU 22.04 does not show QUICK.

#### Version
Mission Planner: 1.3.80
UBUNTU: 22.04
MONO: 6.12.0.182

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
QUAD COPTER

#### Hardware type
SITL

#### Logs
[image]

M-sudo-ct, 2023-05-18:
did you find the solution?
because the same happening with me

M-sudo-ct, 2023-05-18:
I solved that problem, just download the 1.3.76 version of mission planner

assada, 2024-02-23:
same issue on 1.3.81

NanoNM, 2025-08-31:
same issue on 76 ubuntu 24.10

R-247-dev, 2025-12-25:
Any solution for Ubuntu 24.04?