ArduPilot/MissionPlanner #3115: Dronecan set param function sending incorrect value for integer strings

mwrightE38, 2023-05-01:
#### Issue details
Large strings that can equate as an integer number cause the dronecan set parameter function to send in scientific notation.

#### Version
1.3.74 checked on master and it's still present.

#### Platform
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Any
#### Hardware type
Cube orange / bluemark remote ID module. Hardware type doesn't effect the issue.

#### Logs
No logs but in the function here https://github.com/ArduPilot/MissionPlanner/blob/master/ExtLibs/DroneCAN/DroneCAN.cs#L2024

When a string is greater than 12 digits all numeric (probably less than this number. This is the length of the number I was using.) the object converts the number to scientific notation.

meee1, 2023-05-01:
Example number?

meee1, 2023-05-01:
ok found it

https://dotnetfiddle.net/NkezWD