ArduPilot/MissionPlanner #3117: When I use korean characters in plugin, it is not displayed well, it is broken

orangefabri, 2023-05-03:
#### Issue details
With the previous version of Mission Planner, I was able to display Korean characters in the console and GMap marker. However, with the newest version, I am unable to display Korean characters on the Mission Planner screen, such as the message window, console, and GMap marker, when using the Mission Planner plugin. the characters are broken.

Please refer to the following link for more information on the issue:
https://discuss.ardupilot.org/t/broken-characters-in-mission-planner-plugin-function-after-update/98719

#### Version
_What version was the issue encountered with_
Mission Planner 1.3.80 build 1.3.8479.20539

#### Platform
[V] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
N/A

#### Hardware type
N/A

#### Logs
N/A

orangefabri, 2023-05-03:
below is code that regenerate the problem :

using System;

namespace MDTPlugins
{
    public class MDTPlugins : MissionPlanner.Plugin.Plugin
    {     
        public override string Name
        {
            get { return "Regenerate Error 2"; }
        }

        public override string Version
        {
            get { return "0.30"; }
        }

        public override string Author
        {
            get { return "Truth"; }
        }

        public override bool Init()
        {
            return true;
        }

        public override bool Loaded() 
        {
            Console.WriteLine("플러그인 로드");
            loopratehz = 12.0f;

            return true;
        }

        public override bool Loop()
        {
            Console.WriteLine("플러그인 루프");
            loopratehz = 20.0f;
            return true;
        }

        public override bool Exit() 
        {
            return true;
        }
    }
}

orangefabri, 2023-05-03:
below is the result of code :

https://drive.google.com/file/d/1btSeTh5IYsNiPTJ9WANx5GS08DHxuA2o/view?usp=share_link

meee1, 2023-05-03:
What's you default system font?

orangefabri, 2023-05-03:
I tried versions 1.3.79, 1.3.77, and 1.3.76, but none of them were able to display Korean characters. It’s a very mysterious issue.

anyway, After clicking the ‘OK’ button on the update message window, the Korean characters started to appear broken.

orangefabri, 2023-05-03:
> What's you default system font?

thank you for your fast reply. I'm finding the default system font now. please wait

orangefabri, 2023-05-03:
> What's you default system font?

I'm using windows 11, and its default font is "Gulim" or "gulim", as I searched on the web. 

I couldn't find the way to check it on regedit or control panel yet.

orangefabri, 2023-05-03:
> What's you default system font?

in the notepad font setting, I found that font is "Consolas"

orangefabri, 2023-05-03:
> What's you default system font?

for your reference, I could regenerate this issue on my friend's laptop.(mission planner 1.3.80)

orangefabri, 2023-05-03:
> What's you default system font?

for your reference, my friend's laptop(which runs mission planner 1.3.74) korean text displayed well.

orangefabri, 2023-05-03:
> What's you default system font?

for your reference, also in my laptop, mission planner 1.3.74 displays korean text very well.

orangefabri, 2023-05-03:
> > What's you default system font?
> 
> for your reference, also in my laptop, mission planner 1.3.74 displays korean text very well.

1.3.75 and above versions does not display korean characters well.

1.3.74 display korean characters well.

meee1, 2023-05-04:
set
Console.OutputEncoding = Encoding.UTF8;
in your plugin for console

need an exmaple thats not console based

orangefabri, 2023-05-21:
I added "Console.OutputEncoding = Encoding.UTF8;" to my plugin example, but it isn't working.

I want to display most of the information in English, but it would be better to show the landmark names in Korean. Therefore, I am currently working on it.

orangefabri, 2023-05-21:
below example will regenerate error, and not console-based :

using System;
using System.Windows.Forms; 
using GMap.NET.WindowsForms.Markers;
using GMap.NET;
using GMap.NET.WindowsForms;
using System.Drawing;

namespace MDTPlugins
{
    public class MDTPlugins : MissionPlanner.Plugin.Plugin
    {
        private GMapOverlay shipOverlayFP; 
        private GMapOverlay takingOffPositionOverlay; 
        
        public override string Name
        {
            get { return "Regenerate Error"; }
        }

        public override string Version
        {
            get { return "0.30"; }
        }

        public override string Author
        {
            get { return "Truth"; }
        }

        public override bool Init()
        {
            return true;
        }

        public override bool Loaded() 
        {
            takingOffPositionOverlay = new GMapOverlay();
            takingOffPositionOverlay.IsVisibile = true;
            Host.FPGMapControl.Overlays.Add(takingOffPositionOverlay); 

            MarkingTakingOffPositionMarker();

            loopratehz = 12.0f;

            return true;
        }

        public override bool Loop()
        {
            loopratehz = 20.0f;
            return true;
        }

        public override bool Exit() 
        {
            return true;
        }

        private void MarkingTakingOffPositionMarker()
        {
            var location1 = new GMap.NET.PointLatLng(35.0724, 129.05471);
            var marker1 = new GMap.NET.WindowsForms.Markers.GMarkerGoogle(location1, GMap.NET.WindowsForms.Markers.GMarkerGoogleType.blue);
            marker1.ToolTipText = "location 1";
            marker1.ToolTipMode = GMap.NET.WindowsForms.MarkerTooltipMode.Always;
            marker1.ToolTip.Offset = new Point(-50, 10); 
            marker1.ToolTip.Font = new Font("Arial", 7, FontStyle.Bold);
            marker1.Size = new System.Drawing.Size(25, 25);
            takingOffPositionOverlay.Markers.Add(marker1);

            var location2 = new GMap.NET.PointLatLng(35.06711, 129.06562);
            var marker2 = new GMap.NET.WindowsForms.Markers.GMarkerGoogle(location2, GMap.NET.WindowsForms.Markers.GMarkerGoogleType.blue);
            marker2.ToolTipText = "location 2";
            marker2.ToolTipMode = GMap.NET.WindowsForms.MarkerTooltipMode.Always;
            marker2.ToolTip.Offset = new Point(-30, 10); 
            marker2.ToolTip.Font = new Font("Arial", 7, FontStyle.Bold);
            marker2.Size = new System.Drawing.Size(25, 25);
            takingOffPositionOverlay.Markers.Add(marker2);

            var location4 = new GMap.NET.PointLatLng(35.06151, 129.07614);
            var marker4 = new GMap.NET.WindowsForms.Markers.GMarkerGoogle(location4, GMap.NET.WindowsForms.Markers.GMarkerGoogleType.blue);
            marker4.ToolTipText = "감지해변";
            marker4.ToolTipMode = GMap.NET.WindowsForms.MarkerTooltipMode.Always;
            marker4.ToolTip.Offset = new Point(-30, 10); 
            marker4.ToolTip.Font = new Font("Arial", 7, FontStyle.Bold);
            marker4.Size = new System.Drawing.Size(25, 25);
            takingOffPositionOverlay.Markers.Add(marker4);

            var location5 = new GMap.NET.PointLatLng(35.04995, 129.08549);
            var marker5 = new GMap.NET.WindowsForms.Markers.GMarkerGoogle(location5, GMap.NET.WindowsForms.Markers.GMarkerGoogleType.blue);
            marker5.ToolTipText = "location 5";
            marker5.ToolTipMode = GMap.NET.WindowsForms.MarkerTooltipMode.Always;
            marker5.ToolTip.Offset = new Point(-20, 10); 
            marker5.ToolTip.Font = new Font("Arial", 7, FontStyle.Bold);
            marker5.Size = new System.Drawing.Size(25, 25);
            takingOffPositionOverlay.Markers.Add(marker5);

            var location6 = new GMap.NET.PointLatLng(35.07416, 129.08309);
            var marker6 = new GMap.NET.WindowsForms.Markers.GMarkerGoogle(location6, GMap.NET.WindowsForms.Markers.GMarkerGoogleType.blue);
            marker6.ToolTipText = "location 6";
            marker6.ToolTipMode = GMap.NET.WindowsForms.MarkerTooltipMode.Always;
            marker6.ToolTip.Offset = new Point(-50, 10); 
            marker6.ToolTip.Font = new Font("Arial", 7, FontStyle.Bold);
            marker6.Size = new System.Drawing.Size(25, 25);
            takingOffPositionOverlay.Markers.Add(marker6);

            var location7 = new GMap.NET.PointLatLng(35.07823, 129.08888);
            var marker7 = new GMap.NET.WindowsForms.Markers.GMarkerGoogle(location7, GMap.NET.WindowsForms.Markers.GMarkerGoogleType.blue);
            marker7.ToolTipText = "location 7";
            marker7.ToolTipMode = GMap.NET.WindowsForms.MarkerTooltipMode.Always;
            marker7.ToolTip.Offset = new Point(-50, 10);
            marker7.ToolTip.Font = new Font("Arial", 7, FontStyle.Bold);
            marker7.Size = new System.Drawing.Size(25, 25);
            takingOffPositionOverlay.Markers.Add(marker7);
        }

    }//end of Class Plugins
}// end of Namespace Plugins

orangefabri, 2023-05-31:
I believe I solved this problem, but I don’t get a clue why this happens.

I believe this problem is related to file’s setting itself.

let’s suppose that C# file A displays korean well and C# file B displays korean bad.

If I copy C# file A and write the same code above, the korean displays well.

but if I copy C# file B and write the same code above, the korean is not displayed well!

I looked through the file’s properties, but couldn’t find a clue.

I’m using windows 11

orangefabri, 2023-06-01:
I opened the files with Notepad++ and found that the encoding types were EUC-KR and UTF-8 BOM.

I suspect that reinstalling Mission Planner affected the encoding of the CS file, but I have no clue.

Anyway, I solved the problem, so I'm closing this issue with this comment.

closed 2023-06-01 by orangefabri