ArduPilot/MissionPlanner #3119: UI Language change from Korean to another is imposible

Arccoma, 2023-05-10:
[image]

#### Issue details
The UI language of the latest MP installed on Windows 10, where the system language is Korean, is Korean.
I want to change the UI language to English, but I can't select a language in the drop-down option, and the speech option appears incorrectly.

After changing the system language of Windows to English and reinstalling the MP, I can change the UI language of the MP normally. but
Once changed to Korean, it will return to the original state and cannot be changed to another language.

This problem appeared only in Korean among 17 languages supported by MP.

#### Version
Mission Planner 1.3.80

Arccoma, 2023-05-10:
https://github.com/ArduPilot/MissionPlanner/blob/dc11a292b974bab22f3c5b71607d76e896c2d76d/GCSViews/ConfigurationView/ConfigPlanner.ko-KR.resx

I don't understand this code, but I guess there is a problem somewhere in this file?

orangefabri, 2023-05-21:
https://discuss.ardupilot.org/t/broken-characters-in-mission-planner-plugin-function-after-update/98719/13

I solved this problem editing config.xml

but I found some version of mission planner doesn't have language option on config.xml.

and also, there is broken korean character problem on plugin. you can see in the link.

Arccoma, 2023-05-25:
[image]

I compared the differences between ConfigPlanner.resx (Left side) and ConfigPlanner.ko.KR.resx (Right side) using the WinMerge program[](https://winmerge.org/downloads/?lang=en) .

The position coordinates of the Speech combo box in the left file and the UI language combo box in the right file almost match. (There is only a difference as much as y coordinate 1)  
This caused the two combo boxes to overlap, but the bottom line seems to look thick.  
The two combo boxes are the same size, but they have almost the same location coordinates, so only one came out, and depending on the ZOrder value, what is drawn later seems to cover the previous one.  

I think it will solve the problem if I lower all the elements below the UI language one by one, but it will be cumbersome to modify the y coordinates one by one.

So I just copied all the contents of the left file, replaced the entire contents of the right file, changed only the text values of the elements back to Korean, and rebuilt them to solve the problem for now.
[image]

However, I haven't made a Pull Request yet because I haven't figured out exactly when and why this issue occurred and so I'm not sure if this solution is appropriate.

orangefabri, 2023-05-26:
please use ctrl+f

orangefabri, 2023-05-28:
> > I solved this problem editing config.xml
> 
> @orangefabri What is the config.xml file and where can i find it ? And can you tell me specifically how you modified that file to solve the problem?

you can find config.xml in below path :
"documents\Mission Planner\config.xml"

if you can't find the language tag existing, you can add new tag like below :
<language>en</language>

maybe you can find KR, KO, or ko here, with existing tag.

but sometimes you missing that tag in config.xml

and for your reference, there is typo in your title(not 'imposible', 'impossible')

My all korean friends are suffering this problem. 

and even more, sometimes languages automatically changes to korean, and if you are using language korean, you can't even run quadplane in SITL tab.

I want to raise more issues to mission planner developer.

1. Korean characters are broken on plugin functions(https://github.com/ArduPilot/MissionPlanner/issues/3117)

2. If you are using Korean characters, can not run Quadplane SITL(you should use English)

3. sometimes languages return to korean. If you change mission planner's title by adding "logo.txt" in mission planner directory, suddenly there appears "title.xml" in "documents/mission planner", and  languages are set to as defined in "title.xml"