ArduPilot/MissionPlanner #3134: Full Parameter List Name column is slightly too narrow

rmackay9, 2023-06-21:
The enhanced Full Parameter List has some nice features including the ability to set the individual bits of BitMask parameters but the Name column is too narrow meaning the user needs to expand it in order to see the full parameter name.  Below is an example of me attempting to view the gimbal/mount parameters.
[image]

4O6FPV, 2023-07-05:
this also is similar to the user parameters selection list not being large enough to view all the parameters. Issue here https://github.com/ArduPilot/MissionPlanner/issues/3114

rmackay9, 2023-08-03:
This appears to be fixed in beta MP.

xfacta, 2023-08-26:
I would suggest we reopen this as the Name column can not be resized in 
_MP/Config/Full Parameter List_ 
or in 
_Review A Log/Show Parameters_

Other columns can be resized.
I have done Beta updates and currently on 1.3.80 build 1.3.8644.43133

rmackay9, 2023-08-28:
@xfacta,

OK, reopened.

It seems like both the "Name" and "Units" columns cannot be resized.

I guess you've found situations where the "Name" column is not wide enough?  For me it cannot be resized but it is always wide enough.

xfacta, 2023-10-28:
Seems fixed again after a beta update

rmackay9, 2023-10-28:
Yes, it seems fixed, closing!

closed 2023-10-28 by rmackay9