ArduPilot/MissionPlanner #3198: Param loading Green line does not go away

rmackay9, 2023-10-03:
I think there may be an issue with the latest beta which means that the green parameter bar never goes away.

[image]

rmackay9, 2023-11-13:
It seems the "Retry Now" button appears every time a parameter is changed.

EosBandi, 2024-01-06:
Is it still happens with 1.3.81 ? I cannot repro it.

rmackay9, 2024-01-06:
Hi @EosBandi,

Thanks for looking into this.  the exact way that I can reproduce this is to:

1. Start MP's copter simulator with "wipe" checked
2. Open config, Full Parameter list and set MNT1_TYPE = 10 (Xacti)

.. the green line should appear but never go away.

EosBandi, 2024-01-06:
OK, got it. It seems to be a generic issue, when setting a parameter changes the number of parameters.
The quickest way to solve it to notify the user and do a forced parameter refresh.

Can you imagine a scenario when this forced refresh can cause an issue ?

rmackay9, 2024-01-06:
On very slow links downloading all parameters could be very slow and could even timeout.  That's the only real problem I think.  I don't think the autopilot's performance (e.g. the main loop) will slow down during a parameter download.

PR #3271 (2024-03-04): ConfigRawParams: When setting a parameter changes the number of parameters, do a full refresh. and check if reboot is needed
If a param change causing the change in the number of parameters, the parameter subsystem could stuck in a "downloading is in progress" state. This pr:
removes the name check, since not only _enable parameters can change the number of parameters
Add a check at the end of Write Params and if the reported param number not equals with the current number, it notify the user and do a param refresh.
This fixes #3198
closed 2024-03-04 by meee1