ArduPilot/MissionPlanner #3213: Gstreamer: update to most recent version

zeethas, 2023-10-20:
I compared the results of video streaming on a 4G network using the same source stream (a Raspberry Pi + camera) and 2 different viewing clients on ground station (PC), the first client is version 1.14.4 internal to mission planner and the second gstream version is version 1.22 .6 (the most recent to date). 
I noticed that the results with 1.22.6 are much better, in particular there are much fewer artefacts which are normally very visible when the 4G signal is not optimal.
My question is whether it is possible to plan an update of gstreamer internal to mission plammer to the most recent version in the next releases. 
Thank you

#### Version
1.3.80

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
copter

#### Hardware type
Omnibus F4 pro v3

EosBandi, 2024-01-06:
There are still some compatibility issues on windows machines with the latest gstreamer and mission planner.
Normally MP will check if there is an already installed version of gstreamer (minGW) and if found it tries to use. So you can experiment with installing the new version (and delete the MP downloaded version from ProgramData/Mission Planner/gstreamer.
Note: you need the minGW version NOT the MSVC.

closed 2024-01-06 by EosBandi

ArthurPatriot, 2025-08-30:
> There are still some compatibility issues on windows machines with the latest gstreamer and mission planner. Normally MP will check if there is an already installed version of gstreamer (minGW) and [...]

Tested with latest version - it's not working.