ArduPilot/MissionPlanner #3224: translation of mission planner into the Czech language

KSS-LAB, 2023-11-04:
#### Good day, would it be possible to translate the mission planner into Czech please? the link below should contain the necessary files that I generated according to the instructions.

https://drive.google.com/drive/folders/1ox4I7_twNDhuXQqxnRFAKycP4YDFXICD

EosBandi, 2024-01-06:
Hi !
Coudl you please give me access to the files ?

KSS-LAB, 2024-01-12:
hi,

you probably mean the mission planner files? if you tell me how to provide you with these or other files, I will try

Odesláno z iPhonu

> 6. 1. 2024 v 6:22, Andras Schaffer ***@***.***>:
> 
> Coudl you please give me access to the files ?

EosBandi, 2024-01-12:
Yes, Just share them and let me access them. my email is ***@***.***,
I asked permission to the share you write in the issue.
Děkuji,
   Andras

KSS-LAB, 2024-01-12:
I sent you a github email notification to share those generated files for that translation, are they the files you meant?Odesláno z iPhonu12. 1.