ArduPilot/MissionPlanner #3249: "Bad battery" message is misleading

rmackay9, 2023-12-01:
With a Copter running 4.4 (and perhaps other versions) MP displays (and says) "Bad Battery" when the battery voltage is low.  This is likely because AP is setting the [SENSOR_BATTERY](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L199) bit in the [SYS_STATUS message](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L4128C8-L4128C8).

I believe we have a conflict with MAVProxy which uses this bit to change the colour of the battery label (or icon) to red but MP interprets this as a battery failure.

It's really not a failure though at least in the way that the AP flight code interprets it.  So MP voicing "Bad Battery" is just unnecessarily scary.

robertlong13, 2023-12-02:
I absolutely agree that this is unnecessarily scary sounding.

From what I gather, ArduPilot clears the battery health bit when either the battery goes unhealthy (a failure of the battery monitor itself, e.g. loss of SMBus communication), or when a mere battery failsafe has occurred. "Bad battery" isn't the best descriptor for either of those, but unfortunately, whatever error text comes up has to apply to both of them, because we can't easily determine which reason it was.

We could just set the battery icon/text red, but I think a failure of the battery monitor, even briefly, should be a high-priority warning.

EosBandi, 2024-01-02:
I agree with Robert since the health bit in sysstatus is ambiguous there is no short textual description that fits all., and I don't see a possible logic to differentiate between battery failsafe and battery monitoring system error. But if there is a way to decide I'm happy to add the logic to MP.

robertlong13, 2024-01-03:
I can think of three ways to handle this. Not sure which way we want to go, as they all have problems.

1) Check the parameters for the connected aircraft and infer whether the current state would trigger a battery failsafe. Would require maintaining a copy of ArduPilot's battery failsafe logic, which is going to lead to headaches in the future.
2) Parse statustext messages to see if we can get more info. This leads to a "period of uncertainty" while you wait to see if a statustext message comes in which may get dropped over a lossy link.
3) Add a MAVLink message. I think it would be really nice to have a periodically-sent message while you are in failsafe (of any kind) that carries the reason, or multiple reasons, why you are in failsafe.

EosBandi, 2024-01-03:
To make it more complicated, we should handle multiple battery systems as well... 

But if I step back a little, the status bits have the same purpose as the generic annunciator panel on a real plane. It gives a warning that there is something wrong with the given subsystem. Then, the pilot checks other indicators to determine the exact problem and act on it... from this angle of view, a "BAD battery" message is not that bad; perhaps it can be changed to "Battery problem" to sound less frightening.