ArduPilot/MissionPlanner #3264: Decimal vs double

meee1, 2024-01-03:
I've seen an issue where a cast from double to decimal causes an exception. Need to fix mavparam to remove all decimals

An int32 can be 100% represented by a double. So replace decimal with double. Only issue is with float rounding. So needs to be considered

Ie decimal.maxvalue < float.maxvalue

meee1, 2024-01-03:
@CraigElder did you end up getting a tlog for this one?

meee1, 2024-01-03:
[image]

meee1, 2024-01-03:
@[image]

EosBandi, 2024-01-03:
can we assume that if the value is larger that decimal.maxvalue then we can ignore conversion issues and cast it directly to double ? 
my concerns : how can be a parameter value this small/large ? I suspects that flt_max is used somewhere... have to check if float.maxvalue to double then back to float will result the same number. if yes then it is easy, if not then it needs some more logic....