ArduPilot/MissionPlanner #3270: Remove Config's Standard Params and Advanced Params pages

rmackay9, 2024-01-05:
[image]
@Hwurzburg and I think that it would be good to remove the Standrad Params and Advanced Params pages.  I suspect very few people find the format easier to use than the Full Parameter List because:

1. AP flight code doesn't include a good distinction of what is "Standard" vs "Advanced"
2. the page lacks the search feature
3. the Full Parameters page now includes the bit setting feature

IamPete1, 2024-01-06:
I agree.

timtuxworth, 2024-01-06:
I stopped using this screen because I never knew if the parameter I wanted was "Advanced" or "Standard", and it just made it harder to find having to know this first.

EosBandi, 2024-01-06:
1.The whole idea of categorizing Advanced and Standard parameters is coming from the parameter metadata, defined in the source code by the @User field. keeping only the full param tree means that we no longer need that type of distinction ?
2. It does have a search feature (FIND) which search in the descriptions as well, not only in the param name like the full param list.
3. It also have a bit setting feature (using the same metadata as full param list)
4. Numerical Values are ARE min/max checked, while full param list is not.
The only thing which is missing is the Favorites.

Basic and Advanced param screens created to use and 'enforce' boundaries, set in the parameter metadata, do we really want to get rid of it ?

Hwurzburg, 2024-01-06:
yes

EosBandi, 2024-01-06:
OK. I think the most non disruptive way is to redo the layouts 

- Add Full Params list to Basic layout and remove standard/advanced 
- Remove stadard/advanced from Advanced layout
- Keep the current basic and advanced layouts as basic-legacy and advanced-legacy if somebody still wants to use standard/advanced params.

PR #3274 (2024-02-24): DisplayView: Disable ConfigStandardParams and ConfigAdvancedParams 
Disable Standard and Advanced Param view and enables FullParamList in basic and advanced views.
custom layouts not impacted, if needed Advanced and Standard param pages can be enabled.
This fixes #3270
closed 2024-02-24 by meee1