ArduPilot/MissionPlanner #3280: Updateing IronPythons Python Version

GoGoWe, 2024-01-26:
#### Issue details
Currently I am developing an interface that should connect MissionPlanner with another application.
After some research I decided to get the data needed I have to use the integrated Script Engine / IronPython integration. 
Unfortunately the Python Version used is that old that I can not manage to get all packages running with it.

**Is it possible to update IronPython used in the project?** Can I update it by myself and open an pull request?
Or shouldn't I even border getting the PR accepted and find a different way? 

#### Version
1.3.81

EosBandi, 2024-01-26:
The used IronPython library is at 3.4.1 which is the latest version, so there is nothing to update. 
I would recommend to write a plugin instead of Python script.

GoGoWe, 2024-01-26:
Indeed, the newest IronPython library version is 3.4.1. Sorry for the poor research I did before opening this issue. 
Yes I already considered an implementation as plugin, even that I could not convince my team to do so.

Thanks again Andras, maybe I find a hybrid solution and go with the webserver approach you suggested once.

EosBandi, 2024-01-26:
Lack of c# knowledge should not be an excuse from your team :)
Let me know if I can help you with the plugin development.

closed 2024-01-31 by EosBandi