ArduPilot/MissionPlanner #3293: Compass Calibration  on Matek F405-Wing FC

asangag, 2024-02-19:
#### Issue details
I'm using the latest mission planner on windows PC , Matek F405-Wing FC with Arducopter Firmware (stable-4.4.4/MatekF405-Wing) 
HGLRC M100-5883 GPS with Compass. When I tried to calibrate the compass MP does nothing.  GPS is connected to Tx3 and Rx3 (Serial 3) with CL2 and DA2 connected as per the diagram on Matek website (https://www.mateksys.com/?portfolio=f405-wing#tab-id-6) . 

I haven't changed any parameters from the list 
I'm trying to calibrate  it indoors without a GPS lock 
If I flash the INAV firmware onto FC , then I can calibrate Compass without any issues.   

From the screenshot below , I dont see the top compass priority table.  I tried enabling all 3 compass and also with only Compass 1  enabled.  When I click the start button nothing happens. I would assume Mag1 field would show some progress but it stays blank. 

#### Version
[image]

_What version was the issue encountered with_

#### Platform

[  ] Copter

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
 Matek F405-Wing
#### Logs
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2024-02-19:
This is more likely to be a flight code issue or configuration issue rather than a Mission Planner issue.  Could you post in the [Copter-4.4 support forum](https://discuss.ardupilot.org/c/arducopter/copter-44/169) (or use the latest beta and then report in the [Copter-4.5 support forum](https://discuss.ardupilot.org/c/arducopter/copter-45/174)) and if at all possible please include an onboard log (LOG_DISARMED may need to be set to 1 in order to produce a log while disarmed).

asangag, 2024-02-19:
Ok sure . thanks for that.