ArduPilot/MissionPlanner #3294: need FlyTo to ask for the height frame

tridge, 2024-02-21:
The FlyTo dialog box on the map currently assumes GLOBAL_RELATIVE_ALT:
https://github.com/diydrones/MissionPlanner/blob/master/ExtLibs/ArduPilot/Mavlink/MAVLinkInterface.cs#L4344
we need the dialog to ask the user what frame they want to use "Terrain Relative", "Home Relative" or "Absolute" and pass that to the setPositionTargetGlobalInt() message.
I just don't know how to create a dialog like that with a float input and a selection drop down with the 3 choices, I will need some help from a C# developer
Anyone flying BVLOS really needs this change

tridge, 2024-02-21:
the use of the old setWP() interface for ArduPlane should also be removed, all fw supports setPositionTargetGlobalInt()

EosBandi, 2024-02-22:
I can work on it only after mid-March if nobody takes it till that.

robertlong13, 2024-02-22:
I got it. Gonna try to knock it out this weekend.

EosBandi, 2024-10-16:
Ohh, I forgot about this one. Thanks Bob. 
One question, is it possible to replace the old setWP method with  setPositionTargetGlobalInt in this or it is a scope of another PR?

robertlong13, 2024-10-16:
Clearly it got away from me too. [checks calendar] apparently Oct 16th is not "the weekend following Feb 23"

> One question, is it possible to replace the old setWP method with setPositionTargetGlobalInt in this or it is a scope of another PR?

Ah, I missed that bit. Will need to be another PR because Michael has already merged that one. If you're up for it that would be great.