ArduPilot/MissionPlanner #3296: issue in mission planner can anyone tell me how to resolve it

Nani-735, 2024-02-26:
[image]
[image]
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2024-02-26:
Hi @Nani-735,

I don't think mission planner is connected to the vehicle using USB or a telemetry system.  The error written reasonably clearly as it says, "Please connect first!".

I'll close this if that's OK but I'll still see replies.

closed 2024-02-26 by rmackay9

Nani-735, 2024-02-26:
no i have connected usb but i am same error from few days

rmackay9, 2024-02-26:
@Nani-735,

In the picture above, the Connect button is clearly disconnected.

Nani-735, 2024-02-26:
now i have connected the usb but i got same error

rmackay9, 2024-02-26:
Hi @Nani-735,

Which version of ArduPilot is loaded onto the autopilot?  Also which frame type (Copter, Rover, Plane)?  Also can you please reduce the mission to just a single takeoff command and try again?

If that still doesn't work I might ask that you provide an onboard log which can be produced, even while disamred by setting LOG_DISARMED = 1.

I might also ask that you update Mission Planner on your PC by going to the Help screen and pressing the "Check for Updates" button.

By the way, writing waypoints is a very common thing for people to do so we would be overwhelmed with support requests if this was a general problem.

Nani-735, 2024-02-26:
Ok, I'll check it out.

Nani-735, 2024-02-27:
will you say how to do setup settings in mission planner

rmackay9, 2024-02-28:
Hi @Nani-735,

Copter setup is on the wiki here.

If you face issues with the setup it is best to report in the support forums.  [Here's the forum for Copter-4.4](https://discuss.ardupilot.org/c/arducopter/copter-44/169) (the stable version) and [here's the forum for Copter-4.5 ](https://discuss.ardupilot.org/c/arducopter/copter-45/174)(our beta software).

Nani-735, 2024-02-28:
Awesome, thanks for the info!