ArduPilot/MissionPlanner #3298: Tuning/Quick: NAMED_VALUE_FLOAT leads to multiple identical entries

rmackay9, 2024-02-28:
If multiple NAMED_VALUE_FLOAT value are sent from the flight code, each with different names, MP seems to mixup the names and only display one of them on the Action or Tuning tabs list "Display This" screen.  So instead of multiple names appearing, only one name appears multiple times.

This is and example of what the screen shows in this situation:
[image]

EosBandi, 2024-03-05:
Checked on latest beta, no repro. Sent named_floats from a lua script.
[image]
function lrf ()

distance = 100
           gcs:send_named_float("First", distance)
           gcs:send_named_float("Second", distance)
           gcs:send_named_float("Third", distance)
           gcs:send_named_float("Fourth", distance)
           gcs:send_named_float("Fifth", distance)
           gcs:send_named_float("Sixth", distance)
		   
		   
    return lrf, 1000
end

return lrf, 1000

robertlong13, 2024-03-05:
Same, I haven't hit this. @rmackay9 do you have a tlog? Maybe that will help reproduce