ArduPilot/MissionPlanner #3300: Simultaneous Display of Mission, Geofence, and Rally Points in PLAN Screen

tatsuy, 2024-03-01:
I would like to simultaneously display and edit missions, geofences, and rally points in PLAN Screen.
This would enable users to view and edit missions while also creating geofences and rally points in the context of the mission they are planning.

As it currently stands, users must toggle between "MISSION", "FENCE", and "RALLY" options via a dropdown menu located in the upper right corner of the PLAN Screen. This functionality limits the display to only one selected item at a time.

Proposal:
- Maintain the current dropdown menu for selecting the primary editable item ("MISSION", "FENCE", or "RALLY").
- The selected item from the dropdown menu would be displayed in an editable mode, allowing users to add, remove, or modify points as currently possible.
- Items not selected in the dropdown menu would be displayed in a read-only mode. This would allow users to see all elements (mission waypoints, geofence boundaries, and rally points) overlaid on the map simultaneously, but only edit the item type they have selected from the dropdown menu.

By the way, it's noted that previously, the right-click menu in the PLAN screen allowed for simultaneous creation of missions, geofences, and rally points—a feature no longer available when a vehicle is connected, as these options are removed from the menu.
https://github.com/ArduPilot/MissionPlanner/blob/latest/GCSViews/FlightPlanner.cs#L2661-L2673

tatsuy, 2024-03-07:
@EosBandi 
Thank you for your reply.
Indeed, the current functionality that allows fences to be displayed during mission editing is useful. However, it does not permit the simultaneous display of all elements—missions, fences, and rally points—on the fence or rally point screens. It would be beneficial to have the capability to edit fences and rally points while also viewing missions.

jagrandle, 2024-03-07:
I also think this is a great idea. It would be even better if it did not require the vehicle to be connected. Often more complicated missions, which are usually the ones more likely to use all these different types of points or areas, are developed in advance in the office before heading in to the field. Having to be connected to a vehicle is just another level of complexity.
Thanks