ArduPilot/MissionPlanner #3305: Plan: Loiter Radius field should be be used for multicopters

rmackay9, 2024-03-08:
The Plan screen has a "Loiter Radius" field which is probably populated from a parameter if using Plane.  Copter doesn't have this parameter though so we should hide this field and never use it (for multicopters).

This field should also NOT be used (for multicopters) when drawing the circle on the map for LOITER_TURNS commands.  If the LOITER_TURNS command's radius is zero then the circle should not be displayed because the vehicle will do a panorama shot (e.g. a pirouette, e.g. circle in place).

[image]

casrya, 2024-03-08:
@rmackay9 
I took a look at this because I use LOITER_TURNS in missions and had never noticed any problems, and found it is fact working correctly for me with Copter.

I believe it is because my value shown in the disabled loiter radius textfield (which you have highlighted in red) is set to 0.

As a workaround you can try changing it in config.xml as follows (I suspect your existing value will be 80 which you may have set using a different vehicle type?)

<TXT_loiterrad>0</TXT_loiterrad>

P.S I agree this appears to be a vehicle type bug but thought this might be helpful in the meantime?

rmackay9, 2025-01-11:
I think we should disable this field for Rovers as well as discussed on this support forum topic https://discuss.ardupilot.org/t/lidar-collision-avoidance/128497/4