ArduPilot/MissionPlanner #3306: Visual Studio design time issue

HYZ-87, 2024-03-08:
#### Issue details
MethodNotFoundExceptions occur when I tried to open design pages of HUD and FlightData. 
[image]  
I tried [several ways](https://stackoverflow.com/q/78125124/20393761) to solve this, but I still couldn't figure out the reason.  

After I updated the version of VS at the laptop that used to work, I finally found out the main reason that causes this situation.   

The problem occurred after I updated Visual Studio. The design pages worked fine when I used version 17.8.4 until I updated it to version 17.9.2.  

It's either already a potential issue of MP and got exposed due to the bug fix of IDE or it's just the IDE's problem. I can't be certain.  

Since I use the community version, there's no way for me to re-install the old version. Has anyone met this issue before? Is there any workaround?  

#### Version
1.3.80 & commit 1fb0fe26

#### Platform
N/A

#### Airframe type
N/A

#### Hardware type
N/A

#### Logs
N/A

EosBandi, 2024-03-08:
It is a Visual Studio isse, 17.9 broke it. Currently no solution yet, only roll back to 17.8.x

HYZ-87, 2024-03-08:
I see. Thanks for your reply.

robertlong13, 2024-05-15:
I spent the better part of a day trying to tweak the source to bypass this problem (since I don't think Microsoft is going to fix this anytime soon). Unfortunately, I couldn't get anything to work in a way that could be merged into the repo. 

I do, however, have a workaround. Whenever you need to use the designer on FlightData.cs, comment out every line that calls the `GdiGraphics` constructor in HUD.cs:
https://github.com/ArduPilot/MissionPlanner/blob/33aa9ec70649635af77a0388d68e28057f243299/ExtLibs/Controls/HUD.cs#L274
https://github.com/ArduPilot/MissionPlanner/blob/33aa9ec70649635af77a0388d68e28057f243299/ExtLibs/Controls/HUD.cs#L1932
https://github.com/ArduPilot/MissionPlanner/blob/33aa9ec70649635af77a0388d68e28057f243299/ExtLibs/Controls/HUD.cs#L3717

Keep a stash of that change handy.

I have no idea why I can't hide these calls behind a design-mode check. It doesn't seem to work. No matter what I do, those lines attempt to call and throw exceptions, without ever hitting any nearby breakpoints when I do design-time debugging. When I comment them out, my breakpoints correctly hit, and my design-mode checks seem to work. It's baffling.

EosBandi, 2024-05-15:
Indeed it is very strange bug. I spent quite a time as well to get a workaround, without success. As an icing on the cake, VS 17.9.6 and .7 not only gives an error, but crashes when tries open HUD of FlightData in design mode :(

fronders, 2024-07-26:
Hi everyone, 
thought I'd share a solution I came upon on how to downgrade to 17.8.x. if you're stuck on a higher release

I tried rolling back the VS2022 Community edition but the oldest I could get was 17.10.1 (probably the initial one I installed on this PC). 

Then I came across an interesting find [here](https://stackoverflow.com/a/78637995).

So what you need to do is delete the existing install of 17.10.x via the Visual Studio Installer (installed in your system), and then run the downloaded from MS site [VS Community installer](https://visualstudio.microsoft.com/thank-you-downloading-visual-studio/?sku=community&rel=17) supplying it with 17.8.x channel URI and MissionPlanner config from the command line like this:
```sh
visualstudiosetup.exe --channelUri https://aka.ms/vs/17/release.LTSC.17.8/channel --config vs2022.vsconfig
```

Aaaaaaaand voilà! 
It would proceed with the Visual Studio Community 2022 LTSC 17.8 install (current being 17.8.12)

[image]

HYZ-87, 2024-08-20:
Hi, @fronders 
I tried your method and it works, but when I later looked up information about LTSC, I found that it seems to be mostly used for the Enterprise or Professional versions.  Check out [this page](https://learn.microsoft.com/en-us/visualstudio/install/update-visual-studio?view=vs-2022#configure-source-location-of-updates-1).
[image]

Though we used the command line to install, which is different from the method mentioned in the link, I'm worried that this might cause some legal issues. Let me know if I'm overreacting.