ArduPilot/MissionPlanner #3308: MAVFTP: Check CRC after file upload

IamPete1, 2024-03-08:
The FTP protocol has a option to get the CRC32 checksum of a file, this could be fetched after a file upload to confirm it was uploaded correctly. 

This can be currently be done manually:

Grab the CRC from the file:
[image]

[image]

Whe a online tool can be used to get the expected checksum from the local copy: https://www.lddgo.net/en/encrypt/crc

It needs a little custom setup to get the correct result:

[image]

meee1, 2024-03-09:
Unused to do this. The issue is large files where taking to long to CRC, causing other issues

There would need to be a limit on the max size that's crc'd

IamPete1, 2024-03-09:
I guess we could add to the FTP protocol a "check CRC is x" so you would send the expected CRC and then leave the flight controller to deal with what to do if its wrong.

This all came about because of a report of scripts getting corrupted on upload . Its certainly not a common issue.  https://discord.com/channels/674039678562861068/719466647710072832/1215451314700550144

I wonder if there was a bug that has now been fixed or something like that.

hunter-hall, 2024-03-11:
Here is a tlog that should contain the crc error I was getting: https://drive.google.com/file/d/16BGuqd8N30H53Prg8PadIq1kyhnfnpw7/view?usp=sharing

hunter-hall, 2024-03-12:
Update: Upon rolling back to plane v4.4.4 (with my modified lua bindings description file), it works perfectly fine and this error does not exist.

tridge, 2024-03-13:
@hunter-hall i've tested uploading the plane_aerobatics.lua both with MAVProxy and MissionPlanner 1.3.81 on a Pixhawk6X on USB with no errors after 5 uploads