ArduPilot/MissionPlanner #3309: MAVFtp: cannot download @SYS files

rmackay9, 2024-03-12:
[image]
It seems that downloading SYS files using Mission Planner's MAVFtp has stopped working.  I'm unsure if it is an AP change or MP change that is the cause.

FYI @peterbarker who has been working in this area recently.

casrya, 2024-03-12:
@rmackay9 
I don't think this ever worked in SITL did it?

Looking at the ArduPilot source code (latest master) the code for threads.txt etc. is implemented in AP_HAL_ChibiOS/ and as such these can be downloaded using latest MP from a ChibiOS device.

However the calls used by SITL in AP_HAL/ seem to be NO-OPs which is why MP shows file not found.

You can however download storage.bin and crash_dump.bin and other physical files in SITL.

rmackay9, 2024-03-12:
PeterBarker confirms that the issue is just that AP doesn't actually provide these files in SITL.  It seems odd to me that the files are visible but not downloadable but it's an AP flight code issue in any case.

closed 2024-03-12 by rmackay9