ArduPilot/MissionPlanner #3325: Add SysID targeting to mount options dropdown

timtuxworth, 2024-04-09:
#### Issue details
There is now an option for SysID targeting in Mount options, but Mission Planner doesn't yet have an option to select it. Here's the discuss post from @rmackay9 

#### Version
4.5.1 (master)

#### Platform
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Copter, Plane

#### Hardware type
Cube Red, Zealot H743

#### Logs
May be able to add later

rmackay9, 2024-04-09:
Thanks very much.  The full list of mount modes is here https://mavlink.io/en/messages/common.html#MAV_MOUNT_MODE

So we are missing two from the drop-down:

- MAV_MOUNT_MODE_SYSID_TARGET
- MAV_MOUNT_MODE_HOME_TARGET

rmackay9, 2025-01-06:
This drop-down is apparently populated by the MNT1__DEFLT_MODE parameter docs so we just need to update this on the autopilot side and it should appear.  Unfortunately for me at least the drop-down is no longer populated so I'll create a new issue for that.