ArduPilot/MissionPlanner #3331: Log Browser: right-mouse-button menu includes odd "Properties MasterPane" option

rmackay9, 2024-04-15:
[image]

The Log Browser page's graphing window has an few odd looking right-mouse-button menu items:

- Properties MasterPane
- Properties YAxis
- Properties YAxis2

These look like debug and were perhaps accidentally left in place?

rmackay9, 2024-07-05:
Sorry to be a nag but these 3 extra items are a bit annoying.  I don't think it's good that we have developer options like this exposed to the user.

EosBandi, 2024-07-05:
Only Michael knows, the commit was supposed to allow custom Y axis, but it does indeed looks like a debug menu.
I added a PR to remove it, but it is up to Michael. https://github.com/ArduPilot/MissionPlanner/pull/3376

EosBandi, 2024-07-08:
fixed in 18d17d4f936a6a4a5e85a623682cc5b4a6694b62

closed 2024-07-08 by EosBandi

rmackay9, 2024-07-08:
Ah, so happy that those are gone.