ArduPilot/MissionPlanner #3356: 64bit Windows 11 + MP 1.3.81 Can't UDP connect 8266WiFi module (v1.2.3)

lida2003, 2024-05-23:
#### Issue details

1. [On Windows PC] same computer QGC works fine with 8266 module  // Can't be the firewall or windows env issue

2. [On Windows PC] Close QGC and use MP 1.3.81  connect //both port 14550 or 192.168.4.1:14550 failed 

[image]

3. up to date version 8266 1.2.3

[image]

4. status show communication with FC is OK

[image]

5. popup error like below

[image]

6. Pad Mission Planner 1.3.81 works fine

#### Version
1.3.81

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ x ] Rover
[  ] Sub

#### Airframe type
Rover

#### Hardware type
- Acoda-RC H743 Dual

Device name	Daniel-SurfacePro
Processor	Intel(R) Core(TM) i5-7300U CPU @ 2.60GHz   2.71 GHz
Installed RAM	8.00 GB
Device ID	C5FC645F-69FA-4372-A8F8-CB8C596953A3
Product ID	00330-80000-00000-AA147
System type	64-bit operating system, x64-based processor
Pen and touch	Pen and touch support with 10 touch points

Edition	Windows 11 Pro
Version	22H2
Installed on	‎23/‎5/‎31
OS build	22621.3527
Experience	Windows Feature Experience Pack 1000.22700.1003.0

#### Logs
No

meee1, 2024-05-23:
Are you using udpcl?

lida2003, 2024-05-23:
> Are you using udpcl?

Tried, just the same as UDP. Android version works fine with UDP.

lida2003, 2024-05-25:
Doesn't work with 1.2.2 also.

lida2003, 2024-05-26:
There are two Mission Planner now. All checked then issue solved.

[image]

[image]

closed 2024-05-26 by lida2003