ArduPilot/MissionPlanner #3379: NAMED_VALUE_FLOAT revert back to "customfieldx" once Mission Planner restarts

flyrobot27, 2024-07-18:
#### Issue details
When Mission Planner restarts, all the custom fields sent via NAMED_VALUE_FLOAT (Which starts with MAV_) revert back to "customfield", instead of sticking to the MAV_ name

[image]

Even once the drone is connected, the name does not revert back to MAV_:
[image]

#### Version
Mission Planner 1.3.81 build 1.3.8741.25556 ArduCopter V4.5.3 (ee14ff38)

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
N/A

#### Hardware type
Matek H743-WLITE with Raspberry pi 5
Telemetry radio is Holybro SiK Telemetry Radio V3 (915 mhz 500 mv)

#### Logs
I'm not sure what log I can provide but let me know

EosBandi, 2024-07-22:
This is the normal modus operandi since you don't know what names will come from the flight controller at startup.
If you want permanent naming then you have to define the names in the config.xml
<customfield1>MAV_name1</customfield1>
<customfield2>MAV_name2</customfield2>

This is slightly inconvenient, but it allows the system to be dynamic.

closed 2024-07-22 by EosBandi

EosBandi, 2024-07-23:
See mu updated comment above...

flyrobot27, 2024-07-23:
I can see that, thank you

flyrobot27, 2024-07-23:
I also personally found a workaround. Just click to a different panel and click back (I clicked "Config") and it seems the parameter names have refreshed