ArduPilot/MissionPlanner #3387: Dataflash Log Download is extremely slow

kakariki1, 2024-08-05:
#### Issue details
Downloading a Dataflash Log via USB is extremely slow (8 Megabytes takes several hours).

#### Version
1.3.82 (1.3.81 works fine)

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
not relevant

#### Hardware type
SpeedyBee F405-Mini

lcabon258, 2024-08-13:
I have the same problem. My FC is Pixhawk 6C with ArduCopter 4.5.5.

Rodrigo-NH, 2024-10-07:
Same issue here. Described https://discuss.ardupilot.org/t/extremelly-slow-log-download-in-mp-1-3-82/124656/5

Problem is the same despite FC FW version 4.4.4 or 4.5.6

meee1, 2024-10-07:
Please try beta, confirm if it's been fixed already

kakariki1, 2024-10-08:
As far as I can tell the latest beta is working ...

Rodrigo-NH, 2024-10-09:
Issue is fixed for me after applying beta updates

closed 2024-10-09 by robertlong13