ArduPilot/MissionPlanner #3391: MAVFTP: creates endlessly recursive directories after uploading multiple files

rmackay9, 2024-08-16:
[image]
After creating a new APM/scripts/MAVLink directory and dragging-and-dropping three files into it I ended up with this madness where there seemed to be endless recursive directories

meee1, 2024-08-16:
creates? or display issue?

rmackay9, 2024-08-16:
Hi @meee1,

I've just checked the SD card and it did not actually create endless sub directories on the SD card so it appears to just be a display issue.  It also failed to actually copy the files into the new directory

rmackay9, 2024-11-15:
I just want to re-iterate that this is still happening

[image]