ArduPilot/MissionPlanner #3400: Proximity graph scale option

skemp117, 2024-09-04:
#### Issue details
The current proximity window is locked at 5m. It would be nice if that could be set manually or at least default to the PRX1_MAX parameter.

#### Version
Mission Planner 1.3.82 

#### Platform
[X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
quad

#### Hardware type
SITL

#### Logs
N/A

meee1, 2024-09-04:
Press the plus and minus key while the window is open

skemp117, 2024-09-04:
Woah, thanks for the quick reply! Yep that works. What does the MAV size adjust? I tried changing it expecting maybe the spacing of the concentric rings to change, but I don't see anything happening.

Mav size 1
[image]

Mav size 10
[image]

geofrancis, 2024-09-04:
> Press the plus and minus key while the window is open

this is the first time i have ever seen this.

geofrancis, 2024-09-04:
could we add a note somewhere in the proximity page for this, i have been asked a few times.