ArduPilot/MissionPlanner #3405: When the FlightData form opens the designer during secondary development, VS2022ide will flash back.

Qearl-Y, 2024-09-11:
#### Issue details
When the FlightData form opens the designer during secondary development, VS2022 will flash back.
Unable to enter the designer interface for form design

#### Version
1.3.82  

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

robertlong13, 2024-09-11:
Probable duplicate of #3306
See possible remedies there

closed 2024-09-11 by robertlong13

Qearl-Y, 2024-09-12:
> Probable duplicate of #3306
> See possible remedies there

Descriptions found in the application log:
Application: devenv.exe Framework Version: v4.0.30319 Description: The process is terminated due to an unhandled exception. Exception information: System.IO.FileNotFoundException in MissionPlanner.Controls.HUD.Dispose(Boolean) in System.ComponEntModel.Component.Finalise().