ArduPilot/MissionPlanner #3425: Arming Check bitmask guide missing from MP

stevempotter, 2024-10-06:
#### Issue details
1. ARMING_CHECK in Full Parameter List does not say "set bitmask" in the Options column. 
2. When I double-click on that empty Options box, no window pops up with the option checkboxes. Other parameters with bitmasks to set have a pop-up window with checkboxes.
3. There is also no pull-down menu to choose which options.
How are we supposed to set which arming checks to include?

#### Version
MP 1.3.82 build 1.3.8979.17128
Arduplane v. 4.5.6
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Flying wing (Opterra)

#### Hardware type
Matek F765Wing

#### Logs
Not relevant.

EosBandi, 2024-10-06:
Do you have internet connection and write access to c:\ProgramData\Mission Planner folder ?

robertlong13, 2024-10-06:
Could be the issue that was fixed in this recently-merged PR: https://github.com/ArduPilot/MissionPlanner/pull/3246

If not already, try updating to to latest beta (I'm away from home right now and have no way to check if your build number, so sorry if that's already the case)

closed 2024-10-07 by stevempotter