ArduPilot/MissionPlanner #3434: error when building mp1.3.80 based on Visual Studio community 2022

channgan, 2024-10-24:
#### Issue details
[image]
[image]

#### Version
1.3.80

Godeffroy, 2024-10-25:
Did you buid the Extlibs before ? Because you need to build it (at least once) before building mission planner

This one for you : 
[image]

or better the wohle folder 
[image]

channgan, 2024-10-31:
thanks，i copy my friend’s code，maybe the way i use git is wrong

closed 2024-10-31 by channgan