ArduPilot/MissionPlanner #3437: HUD.cs Designer file not working

nabeelqss, 2024-11-04:
#### Issue details
I am really passionte about changing the interface of Mission planner but HUD Designer.cs is not working I want to make some changes to the layout of HUD using some custom design but this file is not opening I tries visual studio community as well as enterprise version of visual studio. Any help and suggestions will be appreciated 
[image]

EosBandi, 2024-11-04:
It is a known big of visual studio, look in issues here. However HUD is a custom control, all items are drawn directly. So Forms Designer can do nothing with its layout.

nabeelqss, 2024-11-04:
Could you help me identify which files I need to update to change the HUD background, sky, and modernize the controls? I’d also like to change the colors and drawings. I want to give it a new look Thanks!

EosBandi, 2024-11-04:
It's all in HUD.cs, but if you have to ask, then there will be more problems... You have to be fluent in c# and openGL

closed 2024-11-04 by EosBandi