ArduPilot/MissionPlanner #3440: Warning from my Android Phone that MP isn't to be trusted.. why?

davidbuzz, 2024-11-10:
[image]
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2024-11-10:
Why, i dont know
[image]

meee1, 2024-11-10:
the version on google play is built by github CI.

i do control what version makes it to the mass's. ie QA

davidbuzz, 2024-11-10:
I'll leave it to you to work out the underlying cause, good luck.