ArduPilot/MissionPlanner #3443: KML overlay fail

LupusTheCanine, 2024-11-11:
#### Issue details
Mission planners fails to load KML as overlay (Map Tools -> KML OVerlay)
[image]

#### Version
Up to 1.3.82 build 1.3.9077.8924 on Windows

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Logs
[Trees.kml.txt](https://github.com/user-attachments/files/17708689/Trees.kml.txt) (remove .txt extension)

EosBandi, 2024-11-11:
MP is quite picky for correctly formatted and simple KLM files.
> Test: Open Google Earth Pro, draw a polygon and save it as a KML. You can load it nicely into MP.

It is very likely that your KML is malformed, or contains elements that is not understand by MP.

LupusTheCanine, 2024-11-11:
> MP is quite picky for correctly formatted and simple KLM files.
> 
> > Test: Open Google Earth Pro, draw a polygon and save it as a KML. You can load it nicely into MP.
> 
> It is very likely that your KML is malformed, or contains elements that is not understand by MP.

Apparently Google Earth Pro produces malformed KMLs by skipping default values at least for colors and line widths.
It would be nice if it handled default color and line widths.

EosBandi, 2024-11-11:
fixed in https://github.com/ArduPilot/MissionPlanner/pull/3444

Changing polygon fill handling behavior could be an issue for those who get used to that all polygon fills are rendered transparent...

LupusTheCanine, 2024-11-12:
> fixed in #3444
> 
> Changing polygon fill handling behavior could be an issue for those who get used to that all polygon fills are rendered transparent...

In my test there was no fill at all. (I tried 40% in Google Earth Pro). I remember polygons filled white (as were all objects in the given KML), wasn't fun.

danilopmaciel, 2025-01-07:
[image]
Same problem here.
Any solution to fix this issue with the custom project filter?

jagrandle, 2025-01-08:
It would be really great if MP was able to handle more features in the KML overlay. This is such a valuable tool for more complicated missions where there might be no go areas, area divisions based on altitude bands or time, points of interest, etc. Even just being able to display different features in different colours, and maybe different line widths, would be an awesome addition. Polygon shading at different opacity etc might also be able to be added later.
Thank you

vinnyferraribr, 2025-03-13:
[image]

Same error here. Old versios of MP open this .kml normaly, the last and the beta cant`t open some .kml files. 

Example of file: 
[kml.zip](https://github.com/user-attachments/files/19231594/kml.zip)

vinnyferraribr, 2025-03-28:
> #### Issue details
> Mission planners fails to load KML as overlay (Map Tools -> KML OVerlay) [image]
>
> #### Version
> Up to 1.3.82 build 1.3.9077.8924 on Windows
>
> #### Platform
> [ X ] All [ ] [...]

Did you solve that? 

I was paying attention, why are you looking for drive D:?

[image]

Could this be linked to the error?

afonsoVale, 2025-04-26:
Also experiencing this on the latest MissionPlanner version. Version 1.3.81 works fine

afonsoVale, 2025-09-17:
Can confirm the issue is still present in version 1.3.83