ArduPilot/MissionPlanner #3462: Data, Action Mount mode blank

rmackay9, 2024-12-21:
For some reason the Data screen's Action Tab's Mount drop-down appears to be blank.  I'm not sure how MP populates this drop-down but if it's from a flight code parameter I'm happy to fix asap.  In any case, it's quite important that users be able to change the mount mode

[image]

robertlong13, 2024-12-24:
This is really weird:
1. I cannot reproduce
2. I know exactly what caused it

It's caused by the rename of `MNT_DEFLT_MODE` to `MNT1_DEFLT_MODE`. What I can't understand is why it's not happening to me.

closed 2024-12-24 by meee1

casrya, 2024-12-26:
FYI there is also a mount mode combo in class Joy_Mount_Mode for the joystick settings.

PR #3464 (2024-12-24): FlightData: update CMB_mountmode population
Fixes #3462

Also adds two modes that were missing because we were looking for the old parameter name.