ArduPilot/MissionPlanner #3463: I change UI，but it doesnt work，what can i do

channgan, 2024-12-21:
[image]
[image]

Godeffroy, 2025-01-20:
Maybe you post what changes you made ? what files did you modify ? what platform do you compil ? what what platform you execute it ?
Be a little more expressive ?

channgan, 2025-01-20:
i delete the image of plane and rover of sitl.cs as i show in figure 1,and it run as figure 2.
SO,why my modification didnt work?

rmackay9, 2025-02-18:
I think this is really a development question so I don't think we should put this on the issues list.  Sorry to be strict but if we let all questions remain on the issues list we will lose track of all bugs and enhancement requests which is what the issues list is really for.

closed 2025-02-18 by rmackay9