ArduPilot/MissionPlanner #3474: Data: Set Mount drop-down is not being populated

rmackay9, 2025-01-06:
It seems (for me at least) that the Data screen's Actions tab's "Set Mount" drop-down is no longer being populated.

[image]

I discussed briefly with @meee1 and he suggested it is most likely an autopilot parameter change and based on [this line in MP](https://github.com/ArduPilot/MissionPlanner/blob/master/GCSViews/FlightData.cs#L2666) I think it must be the MNT1_DEFLT_MODE parameter description but this hasn't changed in 2 years ([see here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_Params.cpp#L20))

I also see that MP is displaying the options in the full parameter list
[image]

It is quite important that users are able to change the mount mode so it would be great if we could figure out what has gone wrong

EosBandi, 2025-01-06:
fixed in https://github.com/ArduPilot/MissionPlanner/pull/3464

robertlong13, 2025-01-06:
Yeah, the version/build number seems to match your duplicate closed issue, Randy. Looks like you're on the latest beta, but it hasn't made it into the beta yet. Could you grab the latest CI artifact off master and confirm it's working for you?