ArduPilot/MissionPlanner #3489: Support for IP Camera Streaming

SoorajR-ai, 2025-02-15:
## Feature Request: Add Support for IP Camera Streaming in Mission Planner

Mission Planner dont have support for IP cameras. This will enable users to view realtime video feeds from IP camera (RTSP/MJPEG) within the HUD.

- Allow users to enter an **IP camera URL** (RTSP/MJPEG) in the video settings.
- A UI option in **Config/Tuning → Planner** for selecting an IP camera.

### References
- Example RTSP URL: `rtsp://username:password@IP_ADDRESS:PORT/stream`

EosBandi, 2025-02-15:
It has support for all type of IP camera. gstream is integrated into Mission Planner so you can open any video stream in the HUD, just right click on the HUD and add your gstream pipeline to open your camera's stream.

rmackay9, 2025-02-15:
There's also a new right-mouse-button menu item called "Video" I think that supports RTSP streams.  I hope to publish a youtube video showing how this works in the near-ish future

SoorajR-ai, 2025-02-15:
There are no RTSP streams available in the "Video" option. I tried using GStreamer but failed multiple times to connect to an RTSP link.

Which file contains the code for video streaming?

EosBandi, 2025-02-15:
rtspsrc location=rtsp://
192.168.0.12/user=admin&password=&channel=1&stream=0.sdp? latency=100 !
queue ! rtph264depay ! h264parse ! avdec_h264 !videoconvert !
video/x-raw,format=BGRA ! appsink name=outsink

just replace the location with your rtsp url.

closed 2025-02-15 by EosBandi

rmackay9, 2025-02-17:
The  intention is that users will install the [video-stream-information.lua](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/applets/video-stream-information.lua) Lua script and then set the VID1_TYPE parameter to the camera they are using and/or fill in some of the other VID1_xxx parameters if it's not a known supported camera