ArduPilot/MissionPlanner #3491: Possible to Flash FW via UDP / Mavlink connection?

Espenf, 2025-02-17:
Hi, I wonder if it's possible to flash FW via Sololink/UDP/MAVLINK to a 3DR Solo drone with SoloCube.
When WIFI Sololink is connected to the drone I can use MP (auto connected via UDP). Tried to Flash FW but could not make the connection to Cube.  Cube has AC 4.5.3 Flashed via USB port but it's a bit tricky to access this port for updates.  
Anyway thanks a lot to the developers!  MP is the best, love it 🥇

rmackay9, 2025-02-17:
Hi @Espenf,

Yes, it should be possible to update via the SD Card and MAV FTP can be used to upload a new firmware to the SD card.  We have a wiki page here and it includes a video at the bottom showing the process https://ardupilot.org/copter/docs/common-install-sdcard.html

I'm going to go ahead and close this but I'll still see replies, thanks!

closed 2025-02-17 by rmackay9

Espenf, 2025-02-19:
Hi @rmackay9,  Tnx a lot for quick reply Randy. I read the docs and saw your video. It seems a bit untested and no FW yet available for ver 4.5.3 and up to 4.5.7 CubeBlack or CubeGreen-solo. Only CubeOrange has this "Ardupilot.abin" listed for download.  I can live with the risk of bricking the CubeBlack when updating the bootloader :)  If there will be a future FW-version for the standard Cube used on Solo  I suggest to name it like "CubeBlack-solo" 
 By the way, I tested flashing my CubeBlack-solo with the CubeBlack ver 4.5.7 via USB about a week ago but it was a problem for me to make it bind with Controller afterwards. This is why I wanted to retry now since I got version 4.5.3 running.  Have a nice day and tnx for your help. Regards Espen

rmackay9, 2025-02-20:
Hi @Espenf,

Thanks for that, I've created an enhancement request for our build server to create the .abin files for more boards.  I can't promise when that'll happen 'cuz it's a bit outside my abilities https://github.com/ArduPilot/ardupilot/issues/29335

Espenf, 2025-02-20:
Hi,
That's good news and I hope they fulfill the request. 
Tnx again, and have a nice evening.
Regards