ArduPilot/MissionPlanner #3492: MAVFTP shows duplicate @SYS folders

rmackay9, 2025-02-18:
[image]

The MAVFTP screen shows the "@SYS" folder twice.  It is not two separate but identically named folders but rather the same folder.

I've confirmed that MAVProxy does not show the folder twice so I think it's an MP issue.  Here's the output from MAVProxy when listing the directories

[image]

This was first reported here during Copter-4.6 beta testing

bulkovskyi, 2026-10-14:
Opened #3823 to fix this. Since ArduPilot 4.6 the root directory listing includes `@SYS`, and the MAVFTP screen was also adding its own `@SYS` node for older firmware, so the folder showed up twice. The PR only adds that node when the root listing doesn't already have it. Tested against Plane SITL.

rmackay9, 2026-10-14:
@bulkovskyi, thanks very much!!

closed 2026-10-15 by meee1