ArduPilot/MissionPlanner #3522: AutoAnaylse

TheOracle2212, 2025-04-09:
Hi there!
I'd like to contribute to Mission Planner. 
I've got an idea to revamp the AutoAnalyse feature on the DataFlash logs screen. I've been using PyMavlink to write a basic program (at the moment!) to start parsing the .BIN files. Because I have no experience with C# development, I'll need some help in integrating the Python program into Mission Planner. 
So far, these are all the features - feel free to suggest some more:
1.  Automatic Sensor discovery
2.  Plotting all desired roll, pitch and, yaw values against the actual roll, pitch and, yaw values (and warning the user that there is a large discrepancy between any two pairs of points)
3.  Generating information if the motor vectors are incorrect. There are two possibilities here. For a Yaw Imbalance, it simply looks at the error logs and rewarns the user if otherwise unaware. For something like "the leans" a simple algorithm works by comparing the values of the AHRS2.Roll, NKF1[0].Roll and NKF1[1].Roll or the AHRS2.Roll, XKF1[0].Roll and XKF1[1].Roll (dependent on the EKF*). 
4. Vibration checks are undergone. Simply, the program finds if at any time there were any vibrations, on any axis - independent and dependent of each other (specified) that were above 30ms-1.  
5. Magnetic interference. The Mag sensors (X,Y and, Z) is compared to the throttle. The program will search through and see if it can find any sort of correlation between the Mag field and the throttle. Ideally, this value should be below 30%
6. All errors are thrown back to the user in warnings (failsafes included) just to remind them
7. Power issues are analysed and evaluated. Brownouts, random voltage spikes, voltage dropouts and, excessive current draw is evaluated. Most of these come when the Baro or EKF altitude last reports the altitude is definitely above 0. 
8. 
The program goes through scanning for each point (1-7) at a time if the logs are there. 

All issues, errors and warnings are reported back to the user based on the severity. Most things would be a warning (if the vehicle can still continue, but with caution). An error would be a return to a safe position (either a land or return to home) immediately and to analyse the situation.

There will be more features, like automatic notch filtering calculation and such (I'm not too sure how that'd work out) that I'd like to implement as well. I think this could become a good tool to help people analyse their vehicle, after a crash (identification of problem), improve their tuning, identify problems they never even knew they had, or help newbies identify their basic problems in an easy-to-follow kind of way. 

Let me know what you think!

TheOracle2212, 2025-04-09:
I've got the program to a stable point so far, that I'm confident is ready for Beta Testing. 

[autoanalyse.txt](https://github.com/user-attachments/files/19664616/autoanalyse.txt)

I've attached it as a .txt, but you have to change the filename to .py (Github restrictions). To run the program, do ```python autoanalyse.py <log file.BIN>```
Linux will require a ```Python3``` invocation.

rmackay9, 2025-04-10:
Hi @TheOracle2212,

I think it might be better to implement any log auto analysis tool as a [WebTool](https://firmware.ardupilot.org/Tools/WebTools/).  [The code repo is here](https://github.aaacom/ArduPilot/WebTools) and I think as a webtool it will be easier to maintain and upgrade and it will also be usable by more users.

TheOracle2212, 2025-04-11:
Thank you for your reply @rmackay9.
I'm not too sure how to develop for JavaScript for a WebTool. I'll see if I can transcribe it to JavaScript using ChatGPT or something and put it up there. However, I got a friend of mine to convert the program to C# so we can implement this directly into Mission Planner. 
Would it not be more useful in Mission Planner to replace the AutoAnalyse with this version? I'm working on a JS version however to host as a WebTool shortly. 

Additionally, I don't see many (if at all!) docs on the WebTools.