ArduPilot/MissionPlanner #3527: Cell Voltages not displayed

adhit-n, 2025-04-29:
Hi,
Please see the screenshot.

[image]

I have connected two 6s SMBus Battery monitors to the FC. And all the parameters are being shown correctly in the HUD as well as in the Status Tab.
The issue that we are facing is, we are able to see only cell voltages of the first monitor BAT1 in the status tab. The cell voltages of BAT2 are not being shown anywhere although Ardupilot firmware reads cell voltages from both the packs. What we are expecting is, the cell voltages from the second monitor should be displayed from Cell 7-Cell 12 or anywhere else.
So where will I be able to see the cell voltages of the second battery pack? Should any changes be initiated in the MissionPlanner or is it something to do with my configuration of MP?

Also we have noticed that important parameters like Cycle Count and SOH are present in the ardupilot firmware and is also being read from the battery. Can someone help me by adding those parameters and committing in the MP repository?

EosBandi, 2025-04-29:
Individual cell voltages are only handled for the first battery. if you need info about others you have to write a plug-in which subscribed to the relevant mavlink message and display the values. Same goes for the other fields you mentioned.

adhit-n, 2025-05-02:
I tried writing a plug-in using your skeleton code. It didn't work even after a lot of tries. Will appreciate if you could help me with this as I'm not familiar with writing code.
Here's what I have written with the help of AI.
https://drive.google.com/file/d/1M7Ir3vIL-CdFnSST1bYc1huGLwH8YIjy/view?usp=sharing
I have compiled and generated the DLL file. The plugin appears on the plugin manager, but my expected results are not obtained.
Hope you can help me with this

EosBandi, 2025-05-05:
Multiple batteries with cell voltage monitoring is not a common scenario. So i don't see a reason to integrate it to Mission Planner. A plugin system is created for scenarios like these. 

Vibe coding and drones does not mix well, the code above is gibberish. You cannot add variables to the cs from a plugin. You have to display the cell values by your own.

closed 2025-05-05 by EosBandi

adhit-n, 2025-05-06:
PS: I'm not a coder. I have fixed the code that you mentioned as gibberish. I am able to display the cell voltages of all the battery monitors separately in a panel.
I work in an organization where we manufacture smart batteries for drones and other applications. A lot of people here in our country, use two batteries connected in series or parallel. Hence we have received a feedback that they want cell voltages from both the battery packs to be displayed on the GCS.
Thanks for the feedback.