ArduPilot/MissionPlanner #3547: Build on visual studio 2022 broken

danilopmaciel, 2025-06-14:
A reconstrução começou às 08:09...
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\ProjNet\ProjNET.csproj restaurado (em 21 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\BSE.Windows.Forms\BSE.Windows.Forms.csproj restaurado (em 26 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\netDxf\netDxf.csproj restaurado (em 25 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\OSDConfigurator\OSDConfigurator.csproj restaurado (em 26 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\AviFile\AviFile.csproj restaurado (em 2 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\BaseClasses\BaseClasses.csproj restaurado (em 269 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\AltitudeAngelWings\AltitudeAngelWings.csproj restaurado (em 15 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Arduino\Arduino.csproj restaurado (em 290 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\alglibnet\alglibnet.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Antenna\MissionPlanner.Antenna.csproj restaurado (em 28 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Onvif\Onvif.csproj restaurado (em 324 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\7zip\7zip.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\ParameterMetaDataGenerator\ParameterMetaDataGenerator.csproj restaurado (em 330 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\MissionPlanner.Drawing.Common\MissionPlanner.Drawing.Common.csproj restaurado (em 3 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Ntrip\Ntrip.csproj restaurado (em 331 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\LibTessDotNet\LibTessDotNet.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\KMLib\KMLib.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Interfaces\Interfaces.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\LEDBulb\Bulb\Bulb.csproj restaurado (em 2 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\ICSharpCode.SharpZipLib\ICSharpCode.SharpZipLib.csproj restaurado (em 0,9 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Installer\Installer.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\MissionPlanner.Drawing\MissionPlanner.Drawing.csproj restaurado (em 12 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\GMap.NET.Core\GMap.NET.Core.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\GMap.NET.WindowsForms\GMap.NET.WindowsForms.csproj restaurado (em 5 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\GMap.NET.Drawing\GMap.NET.Drawing.csproj restaurado (em 5 ms).
C:\Users\danil\Source\Repos\MissionPlanner\wix\wix.csproj restaurado (em 3 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\MetaDataExtractorCSharp240d\MetaDataExtractor.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\md5sum\md5sum.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Mavlink\MAVLink.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\HIL\MissionPlanner.HIL.csproj restaurado (em 14 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\ManagedNativeWifi.Simple\ManagedNativeWifi.Simple.csproj restaurado (em 0,9 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\LibVLC.NET\LibVLC.NET.csproj restaurado (em 0,9 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Maps\MissionPlanner.Maps.csproj restaurado (em 17 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\NMEA2000\NMEA2000.csproj restaurado (em 380 ms).
C:\Users\danil\Source\Repos\MissionPlanner\Plugins\Shortcuts\Shortcuts.csproj restaurado (em 39 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\MavlinkMessagePlugin\MavlinkMessagePlugin.csproj restaurado (em 43 ms).
C:\Users\danil\Source\Repos\MissionPlanner\Plugins\OpenDroneID2\OpenDroneID_Plugin.csproj restaurado (em 38 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\AltitudeAngelWings.Plugin\AltitudeAngelWings.Plugin.csproj restaurado (em 79 ms).
C:\Users\danil\Source\Repos\MissionPlanner\Plugins\FaceMap\FaceMap.csproj restaurado (em 22 ms).
C:\Users\danil\Source\Repos\MissionPlanner\plugins\Dowding\Dowding.csproj restaurado (em 20 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Zeroconf\Zeroconf\Zeroconf.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\GeoUtility\GeoUtility.csproj restaurado (em 0,9 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\MissionPlanner.Gridv2\MissionPlanner.Gridv2.csproj restaurado (em 86 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\ArduPilot\MissionPlanner.ArduPilot.csproj restaurado (em 381 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\GDAL\GDAL.NET.csproj restaurado (em 9 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\DroneCAN\DroneCAN.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\DriverCleanup\DriverCleanup.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\DirectShowLib\DirectShowLib.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Core\Core.csproj restaurado (em 0,9 ms).
C:\Users\danil\Source\Repos\MissionPlanner\MissionPlannerTests\MissionPlannerTests.csproj restaurado (em 93 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\ExtGuided\ExtGuided.csproj restaurado (em 88 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Flasher\Flasher.csproj restaurado (em 93 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Comms\MissionPlanner.Comms.csproj restaurado (em 14 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\System.Drawing.android\System.Drawing.android.csproj restaurado (em 2 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Controls\MissionPlanner.Controls.csproj restaurado (em 22 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Strings\MissionPlanner.Strings.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\SvgNet\SvgNet.csproj restaurado (em 6 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\tlogThumbnailHandler\tlogThumbnailHandler.csproj restaurado (em 17 ms).
C:\Users\danil\Source\Repos\MissionPlanner\MissionPlanner.csproj restaurado (em 110 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\SharpKml\SharpKml.csproj restaurado (em 2 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\SharpAdbClient\SharpAdbClient.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\solo\solo.csproj restaurado (em 8 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Xamarin.Forms.Platform.WinForms\Xamarin.Forms.Platform.WinForms.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\ZedGraph\ZedGraph.csproj restaurado (em 2 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\TestPlugin\TestPlugin.csproj restaurado (em 23 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\WebCamService\WebCamService.csproj restaurado (em 1 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\SimpleGrid\MissionPlanner.SimpleGrid.csproj restaurado (em 19 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\px4uploader\px4uploader.csproj restaurado (em 15 ms).
C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\WebAPIs\MissionPlanner.WebAPIs.csproj restaurado (em 13 ms).
[... 342 lines cut ...]
69>CSC : error CS0006: Arquivo de origem "C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Comms\bin\Release\netstandard2.0\MissionPlanner.Comms.dll" não pode ser encontrado
69>CSC : error CS0006: Arquivo de origem "C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Controls\bin\Release\net472\MissionPlanner.Controls.dll" não pode ser encontrado
69>CSC : error CS0006: Arquivo de origem "C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\MissionPlanner.Drawing\bin\Release\netstandard2.0\MissionPlanner.Drawing.dll" não pode ser encontrado
69>CSC : error CS0006: Arquivo de origem "C:\Users\danil\Source\Repos\MissionPlanner\bin\Release\net461\MissionPlanner.exe" não pode ser encontrado
69>CSC : error CS0006: Arquivo de origem "C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\HIL\bin\Release\netstandard2.0\MissionPlanner.HIL.dll" não pode ser encontrado
[... cut ...]
69>CSC : error CS0006: Arquivo de origem "C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\ZedGraph\bin\Release\net472\ZedGraph.dll" não pode ser encontrado
69>CSC : error CS0006: Arquivo de origem "C:\Users\danil\Source\Repos\MissionPlanner\ExtLibs\Zeroconf\Zeroconf\bin\Release\netstandard2.0\Zeroconf.dll" não pode ser encontrado
69>Projeto de compilação pronto "AltitudeAngelWings.Plugin.csproj" -- FALHA.
========== Recompilar Tudo: 4 bem-sucedido, 9 falhou, 68 ignorado ===========
========== Recriar concluído às 08:09 e levou 06,235 segundos ==========

danilopmaciel, 2025-06-17:
[image]flightdata.cs crash

EosBandi, 2025-06-17:
Known issue https://github.com/ArduPilot/MissionPlanner/issues/3306

closed 2025-06-17 by EosBandi