ArduPilot/MissionPlanner #3566: Mission Planner Beta's Simulator can't start AP-4.7.0-dev?

rmackay9, 2025-07-26:
It seems that Mission Planner's simulator can't start using ArduPilot-4.7.0 (latest).  I've seen it fail on at least two different people's PC's with an error about missing cygintl-8.dll and cygiconv-2.dll dlls

[image]

[image]

tpwrules, 2025-07-26:
The cygwin build CI job artifacts get transferred through some mechanism I don't quite understand to [here](https://firmware.ardupilot.org/Tools/MissionPlanner/sitl/) for Mission Planner to download. The DLLs are present there.

Searching the source code for DLLs that can be found, I would guess this list needs updating: https://github.com/ArduPilot/MissionPlanner/blob/a5a00689a2a651be53ac0b56c6c7a1f4cac05726/GCSViews/SITL.cs#L429

There are also DLLs in that directory and that list that haven't been changed in three years and are presumably associated with an older cygwin. It shouldn't be hard to transition to a manifest list, it would be generated [here](https://github.com/ArduPilot/ardupilot/blob/26920e3bcece5d143c3c80f52ce617eec35c30de/Tools/scripts/cygwin_build.sh#L48) .

rmackay9, 2025-07-29:
this is fixed in the latest beta, thanks!

closed 2025-07-29 by rmackay9